ARM-software/vscode-embedded-debug #1: Can't flash using ST-Linkv2

fredo514, 2022-12-06:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows_NT x64 10.0.19044
- VS Code Version: 1.73.1
- Extension Version: v0.3.6
- Target Device: STM32F415RGTx
- Other extensions you installed (and if the issue persists after disabling them): Keil Studio Pack
- When starting a debug session with a ST-Link/v2 connected, I get 
> ERROR: Cannot find USB device with S/N 51ff71068370484820270767, it may be already in use, need a firmware update, or need reconnecting

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I expect the debugger to start

**Code sample and logs**
- Code sample
- `launch.json`
```
{
 "version": "0.2.0",
 "configurations": [
     {
         "name": "FRDM-K32L3A6 (CM4)",
         "type": "embedded-debug",
         "request": "launch",
         "serialNumber": "${command:device-manager.getSerialNumber}",
         "deviceName": "STM32F415RGTx",
         "processorName": "cm4",
         "program": "${command:cmsis-csolution.getBinaryFile}",
         "cmsisPack": "Keil::STM32F4xx_DFP@2.15.0"
     }
 ]
}
```

**Screenshots**

**Additional context**

thegecko, 2022-12-06:
Are you using a debug probe or an on-board debugger on a development board?

fredo514, 2022-12-06:
I don't know if it's related, but I can't flash either. I'm getting:
> Device not found
> fail

Using - `tasks.json`
```
{
	"version": "2.0.0",
	"tasks": [
		{
			"type": "embedded-debug.flash",
			"serialNumber": "${command:device-manager.getSerialNumber}",
			"program": "${command:cmsis-csolution.getBinaryFile}",
			"cmsisPack": "Keil::STM32F4xx_DFP@2.15.0",
			"deviceName": "STM32F415RGTx",
			"label": "embedded-debug.flash: Flash Device",
			"processorName": "cm4",
             "group": {
                 "kind": "build",
                 "isDefault": true
               }
		},
		{
			"type": "cmsis-csolution.build",
			"project": "${command:cmsis-csolution.getCprjPath}",
			"problemMatcher": [],
			"label": "cmsis-csolution.build: Build",
			"group": {
				"kind": "build",
				"isDefault": true
			}
		}
	]
}
```

thegecko, 2022-12-06:
Thanks, this is a known issue with external debug probes and a fix has been created. I'll update here when this is released, which should hopefully be in the next day or so.

thegecko, 2022-12-08:
@fredo514 We have a new release of the embedded debugger (0.4.0) which should resolve your issue. Please let us know if it has.

thegecko, 2022-12-08:
Thanks @fredo514 . These both look like different issues.

Can you confirm you are building the application in debug mode?

@mcgordonite it looks like the `cmsis-csolution.getBinaryFile` command isn't returning the path correctly, would you mind investigating?

fredo514, 2022-12-08:
Right, I assumed that since the .axf file was generated, that it was built in debug mode. I added -g to the compiler options in the csolution.yaml file and re-built.

It got further, but now it complains about memory content mismatch:
```
Device Selected: 51ff71068370484820270767
Parse debug description...
Create system description...
Connect to debug unit (ST-Link)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Parse dbgconf file...
DSQ ERROR: Undefined identifier - 'DbgMCU_CR'
DSQ ERROR: Undefined identifier - 'DbgMCU_APB1_Fz'
DSQ ERROR: Undefined identifier - 'DbgMCU_APB2_Fz'
Connect ('haltOnConnect') to device STM32F415RGTx (cm4)...
Device connected
Halt...
Device halted after connect
Reset ('auto')...
Reset completed
Load application...
Verify loaded image...
	Code section: 0x08000000 - 0x08001727
Program RAM...
ERROR: Contents mismatch at: 0x8000000  (Target=0x40  Required=0x78) !
ERROR: Contents mismatch at: 0x8000001  (Target=0x16  Required=0x6) !
ERROR: Contents mismatch at: 0x8000004  (Target=0xd1  Required=0x29) !
ERROR: Contents mismatch at: 0x8000005  (Target=0x3  Required=0x2) !
ERROR: Contents mismatch at: 0x8000008  (Target=0xaf  Required=0x69) !
ERROR: Contents mismatch at: 0x8000009  (Target=0x4  Required=0x14) !
ERROR: Contents mismatch at: 0x800000c  (Target=0xb1  Required=0x1) !
ERROR: Contents mismatch at: 0x800000d  (Target=0x4  Required=0x14) !
ERROR: Contents mismatch at: 0x8000010  (Target=0xb3  Required=0x65) !
ERROR: Contents mismatch at: 0x8000011  (Target=0x4  Required=0x14) !
ERROR: Contents mismatch at: 0x8000014  (Target=0xb5  Required=0x21) !
ERROR: Too many errors to display !
Verify done
ERROR: Contents mismatch at: 0x8000000  (Target=0x40  Required=0x78) !
ERROR: Too many errors to display !
ERROR: RAM verification failed
ERROR: Failed to program flash
Skipped reset & run target
Go...
```

thegecko, 2022-12-08:
That's because you need to flash the binary first.  You can create a flash task and specify it as a `preLaunchTask` in your launch.json config

fredo514, 2022-12-08:
That worked, thanks! I guess I have some brush up to do on VS Code tasks and stuff...

Note that the filename issue is still there, I an changing it manually here.

thegecko, 2022-12-09:
I've created a new issue for the path problem at:

https://github.com/ARM-software/vscode-cmsis-csolution/issues/12

I think this completes this issue.

closed 2022-12-09 by thegecko