ARM-software/vscode-embedded-debug #2: Restarting debug session tries to reprogram when a flash prelaunchtask is set

fredo514, 2022-12-08:
Type: Bug Report

Restarting the debug session triggers the launch task which tries to re-flash the device when a flash prelaunchtask is set in the launch.json task. It fails because a debug session is already attached to it.

OS and Version: Windows_NT x64 10.0.19044
VS Code Version: 1.73.1
Extension Version: v0.4.0
Target Device: STM32F415RGTx
Other extensions you installed (and if the issue persists after disabling them): Keil Studio Pack

To Reproduce:
Add a flash prelaunchtask (e.g. `"preLaunchTask": "embedded-debug.flash: Flash Device"` to the launch.json task
Click on Debug
Click restart
See the error popup

Expected behavior:
I expect the relaunch button to simply send a reset command to the debugger.

thegecko, 2022-12-09:
Unfortunately this is how the tasks and debug system work in VS Code.

Where possible we are trying to hook into the tasks system to allow developers maximum flexibility, however there just isn't a way to say `only execute this task when a certain condition is met` (e.g. the device needs flashing).

We are considering adding a `flash before debug` debug configuration setting which would give us the ability to track whether a flash is required, in the meantime I would recommend in your case not using `preLaunchTask` and running the flash task manually as you make changes.

Another option is to link the build and flash tasks by adding a `dependsOn` entry to your flash task to execute build before flashing.

cc @jreineckearm

fredo514, 2022-12-09:
Thanks, I'll try it.

I think having the restart button only reset the debugger (i.e reset the program counter, etc) is an important feature. Not only that's what users expect when they send a Reset command on other debuggers, but it would make the workflow faster.

thegecko, 2022-12-09:
> I think having the restart button only reset the debugger

You mean this?

[image]

If so, it resets the debugger but will also re-run any `preLaunchTask`. I would argue this isn't the best behaviour, perhaps an issue could be raised with vscode? https://github.com/microsoft/vscode/issues

there is a related (but opposite) discussion here: https://github.com/microsoft/vscode/issues/102296

fredo514, 2022-12-12:
Right, we might be running into limitations of VS Code, as you mention.

From https://github.com/microsoft/vscode/issues/136384, it looks like some restart-related tasks could get exposed in the future. Anyway, I created https://github.com/microsoft/vscode/issues/168881 to see what they think.

The `dependsOn` option is nice, but it would be even better if it only rebuilt if needed (i.e. if the code has changed).

fredo514, 2022-12-12:
Just an update that VS Code team responded to the ticket. They said that the feature can already be done when the debug adapter sends the `supportsRestartRequest` capability.

thegecko, 2022-12-12:
Hmm, this adapter does support that

closed 2022-12-18 by thegecko