ARM-software/vscode-embedded-debug #3: Debug on Keil Simulator

fredo514, 2022-12-08:
Type: Feature Request

I don't know if it's possible, but it would be nice to be able to run a debug session using the Keil Simulator.

thegecko, 2022-12-09:
> run a debug session using the Keil Simulator

Do you mean executing and debugging a model rather than physical hardware?

fredo514, 2022-12-09:
Yes,

Actually, the full story is that I want to be able to debug off-target (simulator is nice, but a PC executable would work too), so that I can run unit tests (e.g. using Unity) quickly and easily.

thegecko, 2022-12-09:
Thanks,

this is currently possible in our hosted version of our extensions at https://studio.keil.arm.com/ and we are considering bringing virtual hardware model support to VS Code.

I'll update here when we can reveal more!