ARM-software/vscode-embedded-debug #7: Codespaces/Devcontainer support

riestmo, 2023-01-10:
Type: Feature Request

I would like to use the vscode-embedded-debug extension running in Github Codespaces or locally using a devcontainer. So far I was not able to flash or debug my application in such a scenario. Are you planning to support the Codespace/Devcontainer use case for this extension (and possibly all other Arm extensions as well)?

thegecko, 2023-01-10:
Hi @riestmo
This extension has been designed to work in both the scenarios you outline.

In Codespaces, the extension (and the vscode-device-manager extension) need to be installed in the UI extension host by installing them manually (search for arm device manager and arm embedded debug in the extensions panel). Unfortunately, the extension devcontainer entries get installed in the wrong host due to the bug you commented on:

https://github.com/orgs/community/discussions/41636#discussioncomment-4632157

When running in the UI extension host on codespaces, communication to the usb device will be via WebUSB.

Running a devcontainer locally requires usb passthrough on the container (as the extensions use native usb on the desktop). I believe this can be done using privileged mode containers, but be aware that USB passthrough doesn't work on MacOS with docker:

https://github.com/moby/hyperkit/issues/149

But is apparently possible if you use VirtualBox:

https://osxdaily.com/2022/07/02/how-to-connect-usb-devices-to-macos-virtualbox-vm/

riestmo, 2023-01-11:
Hi @thegecko,
I already tried to install the arm embedded debug extension in the UI extension host. However, in contrast to the arm device manager extension, VSCode only allows me to install the embedded debug extension per default in the Codespace (see screenshot) [image]. 

I had to enforce to install the extension in the UI extension host using the following setting

`
 remote.extensionKind: {
        arm.embedded-debug: [
            "ui"
        ]
    }
`

Is this intended? I would expect the same behavior as for the arm device manager.

thegecko, 2023-01-13:
A fix for the target `extensionHost` will be in the next release which should mean you no longer need the host override.

Regarding the failure, the system cannot find the program you specified, can you share your `launch.json` config?

riestmo, 2023-01-13:
Great to hear that the fix for the extensionHost will be in the next release.

The task.json for flashing the hex file looks like this:

`{
	"version": "2.0.0",
	"tasks": [
        {
            "type": "embedded-debug.daplink-flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${command:embedded-debug.getBinaryFile}",
            "problemMatcher": [],
            "label": "embedded-debug.daplink-flash: Flash Device (DAPLink)"
        },
    ]
}`

Running it in VSCode locally, it is able to find the hex file in the workspace. However, as soon as I switch to the Codespace/Devcontainer setup, it throws me the previously mentioned error.

thegecko, 2023-01-13:
You can try putting in the full path in the `program` entry, e,g,:

```json
    "program": "${workspaceFolder}/myapp.hex" 
```

If that fixes it, you can also create a task to just echo out the path for the `getBinaryCommand`: 

```
{
    "label": "Echo",
    "type": "shell",
    "command": "echo ${command:embedded-debug.getBinaryFile}",
    "problemMatcher": []
}
```

In your scenario, I believe you may also be able to flash the file directly by right-clicking your hex file in the explorer.

riestmo, 2023-01-16:
Unfortunately neither of the suggested workarounds were successful. I created a repo for reproducing the extension behavior: https://github.com/riestmo/debug-extension-test. Could you try to reproduce the issue?

Is it possible that the extension tries to find the file locally (since it's installed in the UI extension host) instead of the finding it inside the workspace?

thegecko, 2023-01-16:
Thanks, I can recreate the issue and suspect it is a problem with the schemes used in codespaces. I'll investigate...

thegecko, 2023-01-17:
Thanks for the info @riestmo.

A fix has been published and flashing now works with a codespace of your example repository.

riestmo, 2023-01-18:
Thanks a lot @thegecko! I can confirm that flashing hex files from codespace is working now.

closed 2023-01-18 by riestmo