ARM-software/vscode-embedded-debug #8: Ignoring "default" field of pdcs file for selection of flash algorithms

riestmo, 2023-01-18:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 20.04
- VS Code Version: 1.74.3
- Extension Version: v0.6.2
- Target Device: CY8C624ABZI-S2D44:Cortex-M4

When using the embedded-debug.flash command, the extension is loading a list of flash algorithms from the cmsisPack. As an example, the following flash algorithms are loaded for the Infineon::PSoC6_DFP@1.4.0 device family pack:

[image]

However, this is the entire list of available flash algorithms for this cmsisPack. For working correctly, only the default flash algorithms should be used. These are marked as "default=1" inside the pdsc file (e.g. https://github.com/Infineon/cmsis-packs/blob/master/PSoC6_DFP/Infineon.PSoC6_DFP.pdsc, DsubFamily="CY8C62xA").

As reference for the correct/expected behavior, Keil uVision is only using the default flash algorithms as listed here for the same cmsisPack:
[image]

**Expected behavior**
Only the default flash algorithms are used by the embedded-debug.flash command.

thegecko, 2023-02-07:
Thanks for raising this issue @riestmo. Would you mind looking at this @jreineckearm ?

riestmo, 2023-03-02:
Hi @thegecko & @jreineckearm, any updates at this issue?

jreineckearm, 2023-03-03:
@riestmo , my apologies, this one slipped through. Your expectation about only loading default algorithms is correct. We will investigate this as soon as we can.
Would you mind sharing your task/debug configuration for the investigation as well?

riestmo, 2023-03-03:
Thanks for your reply @jreineckearm!

My task configuration looks like this:

`{
	"version": "2.0.0",
	"tasks": [
		{
			"type": "embedded-debug.flash",
			"serialNumber": "${command:device-manager.getSerialNumber}",
			"program": "${command:embedded-debug.getApplicationFile}",
			"cmsisPack": "https://github.com/Infineon/cmsis-packs/blob/master/PSoC6_DFP/Infineon.PSoC6_DFP.1.4.0.pack",
			"processorName": "Cortex-M4",
			"problemMatcher": [],
			"label": "embedded-debug.flash: Flash Device",
			"group": {
				"kind": "build",
				"isDefault": true
			}
		}
	]
}`

jreineckearm, 2023-03-07:
Thanks @riestmo for providing the task config!
We were able to identify the issue in the debugger and are working on a fix.
We will let you know once this goes into a new release.