ARM-software/vscode-embedded-debug #9: Path error when debugger tries to open source file in Codespaces

riestmo, 2023-01-18:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 20.04
- VS Code Version: 1.74.3
- Extension Version: v0.6.2
- Target Device: CY8C624ABZI-S2D44:Cortex-M4

When starting a debug session in Github Codespaces, the debugger starts successfully and runs to main using the default setting for "debugFrom". However, it fails to open the main.c file as shown below.

[image]

I suspect it's a similar scheme issue as discussed in https://github.com/ARM-software/vscode-embedded-debug/issues/7#issuecomment-1383865800

**Expected behavior**
The debugger opens the C file once a breakpoint is hit.

thegecko, 2023-01-19:
I'll investigate.

Out of interest, are you building the binary in the codespace or somewhere else?

riestmo, 2023-01-19:
I'm building it in the Codespace.

thegecko, 2023-01-20:
This is very frustrating. I think it's a bug in codespaces and github.dev regarding their file schemes.

This used to work, so is a recent regression. I'll need to create a reproducer and file an issue with GitHub

thegecko, 2023-01-20:
Raised here:

https://github.com/orgs/community/discussions/44904

riestmo, 2023-01-23:
Thanks for raising this issue!

thegecko, 2023-04-03:
@riestmo We've released an update to the debug extension this morning which could fix this issue.

Could you confirm if you see any improvement?

thegecko, 2023-04-03:
Nice! Sorry about the delay