ARM-software/vscode-embedded-debug #11: Can't see current line when stopping

fredo514, 2023-02-02:
**Type: Bug Report**

- OS and Version: Windows 10
- VS Code Version: 1.74.3
- Extension Version: 0.6.3
- Target Device: STM32F415RGT6

When running a debug session, I can't see the current line when the debugger is stopped. The output window tells me it's halted on a breakpoint, but the source file line is not highlighted.

**Steps to reproduce the behavior:**
1. Launch debug session on a build compiled with the `debug: "on"` flag.
2. Place a breakpoint somewhere
3. Run until the breakpoint
4. Observe that the source line is not highlighted.

I expected the debugger to highlight the line source line it has stopped on.

[image]

thegecko, 2023-02-03:
Thanks for the feedback @fredo514 

Can you confirm the compiler you used to create your binary and exactly what is meant by `debug: "on"`? Is it a set of optimisation flags?

fredo514, 2023-02-03:
I'm using arm-none-eabi v11.2.1, with the gcc.cmake, installed per the instructions at https://github.com/ARM-software/vscode-cmsis-csolution.

`debug: "on"` is related to https://github.com/Open-CMSIS-Pack/devtools/issues/709#issuecomment-1412734658. It's an option I had to enable manually in the csolution.yaml to get the debugger to load without an "Unable to find symbols" error when I try to debug, unless I explicitly add the "-g" flag to the target compiler options.

thegecko, 2023-02-03:
Thanks for the further information.

@jreineckearm would this issue (and that seen in #10 ) be indicative of further elf loader support needed for GCC?

jreineckearm, 2023-02-03:
I think both problems (this issue and #10 ) relate to the same root cause. Which is that the current PC cannot be associated with a source location.

It would be interesting to have a reproducer for this behaviour to understand if more elf loader support is required. Is this a publicly available project or is it based on one which can be easily modified to provoke the issue?

thegecko, 2023-02-03:
> Is this a publicly available project or is it based on one which can be easily modified to provoke the issue?

@fredo514 would you be happy to share the source (or a binary if not?)

fredo514, 2023-02-03:
It's a simple blinky project converted from a CubeMX generated Keil project. I attached the source. 
[blinky_clicker.zip](https://github.com/ARM-software/vscode-embedded-debug/files/10580131/blinky_clicker.zip)

jreineckearm, 2023-02-06:
Thanks for the reproducer, @fredo514 !

The built ELF/AXF file looks OK from a debug information point of view.

But the address at which the CPU halted ( `0x080019D0` ) is not where you set the user breakpoint. It is a `BKPT` instruction compiled into the system library function `_sys_open` for which no source code or more detailed debug information is available. If run without an attached debugger, this would cause a HardFault.

Hitting the `BKPT` instruction in this particular system library function is typically a result of using `printf` without retarget code. See for example https://www.keil.com/pack/doc/compiler/RetargetIO/html/Retarget_Overview.html .

We will further analyze your report to identify how to make it more obvious that the CPU didn't hit the expected breakpoint. Enhancements to the Call Stack view may already help with this.