ARM-software/vscode-embedded-debug #12: I can setup break points in the app

cekuran, 2023-02-20:
OS: Ubuntu 22.04.1 LTS Jammy
VS Code: 1.75.1
Extension: v0.8.0

I am putting a breakpoint in the source code but when I start to debug it switch the break point to "Unverified Breakpoint"

thegecko, 2023-02-20:
An unverified breakpoint often means it can't map the source to the binary. Is your `binary.out` built with debugging symbols?

cekuran, 2023-02-20:
I am using -g -gdb and -O0
[image]

thegecko, 2023-02-20:
Could you try using the `-gdwarf-4` flag instead of `-ggdb`?

cekuran, 2023-02-20:
I have done it, but its behavior is the same

thegecko, 2023-02-20:
Are you happy to share your binary so we can investigate?

cekuran, 2023-02-21:
Here is the binary

[binary.zip](https://github.com/ARM-software/vscode-embedded-debug/files/10788460/binary.zip)

thegecko, 2023-02-22:
Thanks @cekuran , @jreineckearm would you mind taking a look?