ARM-software/vscode-embedded-debug #13: Not able to flash when using GCC

VVESTM, 2023-02-24:
Type: Bug Report

Wrong file is selected when working with GCC
- OS and Version: Windows
- VS Code Version: 1.75.1
- Extension Version: v0.8.0
- Target Device: STM32 NUCLEO-U575ZI-Q

- I am able to compile with GCC but when flashing and debugging, wrong file is selected (.axf instead of .elf)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'CMSIS extension', select the .csolution.yml file
2. Select "Build Type" GCC
3. Build, this generates the .elf
5. Flash. In this case, I have an error because it is looking for .axf instead of .elf.

For information, here is my task.json:
[image]
Then, in settings, I have:
[image]
So, seems .elf is supported.
(With AC6, I am able to compile, flash and debug with same settings)

Thanks for your help

thegecko, 2023-02-24:
The `getApplicationFile` command used in your task defers to the `cmsis` extension when that's installed (which looks to be the case) and the `Application Pattern` is ignored.

@mcgordonite Can you confirm whether the `cmsis-csolution.getBinaryFile` command returns an `axf` file when using gcc? It may need to return a different path.

In the meantime, this can be worked around by using the full file path to your binary file in the task and not using the command.

VVESTM, 2023-02-24:
Thanks, yes, I am not blocked as I can use the full path (already tested and working). 
But in order to share a single task.json file between several projects, the getBinaryFile would be more convenient.

Can also be linked to https://github.com/ARM-software/vscode-cmsis-csolution/issues/20.

thegecko, 2023-06-21:
@mcgordonite could you confirm if this should work?

mcgordonite, 2023-06-23:
This should now work: the binary file path is determined dynamically based on the generated cbuild.yml file. You should now be able to call your binary whatever you like.

See https://github.com/ARM-software/vscode-cmsis-csolution/issues/20.

VVESTM, 2023-06-23:
close by  https://github.com/ARM-software/vscode-cmsis-csolution/issues/20

closed 2023-06-23 by VVESTM