ARM-software/vscode-embedded-debug #15: Support flashing of Intel Hex programmes

microbit-carlos, 2023-03-20:
Type: Feature Request

Pointing to an Intel Hex file shows this error:
```
Not an ELF file: invalid magic number
Flash failed
```
So I guess it only ELF files are currently support?

It'd be great if flashing Intel Hex files could be supported as well.

thegecko, 2023-03-20:
@microbit-carlos can you try the `embedded-debug.daplink-flash` task type for your use case instead?

You may also be able to right-click the hex file in your workspace and flash it directly if you are using the `arm.device-manager` extension and have a DAPLink device selected.

thegecko, 2023-03-20:
@Elise-Kaminski we need to outline this flash type in the docs for DAPLink devices.

For completeness, the `daplink-flash` task can be added through the VS Code task templating system and should work on vscode.dev/github.dev as well as codespaces.

closed 2023-06-21 by thegecko