ARM-software/vscode-embedded-debug #16: embedded-debug.flash won't work (ERROR: Sequence 'DebugPortSetup' failed) (RDDI-ERROR 13)

kubichris, 2023-04-13:
Type: Bug Report

**Describe the bug**
- VS Code Version: 1.77.3 (user setup)
  Electron: 19.1.11
  Chromium: 102.0.5005.196
  Node.js: 16.14.2
  V8: 10.2.154.26-electron.0
  OS: Win11, Windows_NT x64 10.0.22621
- Extension Version: 0.9.0
- Target Device: J-Link, CMSIS-DAP, ST-LINK progrmmers 

task.json:
```
{
            "label": "embedded-debug.flash: Flash Device",
            "type": "embedded-debug.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${workspaceFolder}/Release/reader.axf",
            "cmsisPack": "${workspaceFolder}/Pack/NXP.PN7xxxxx.0.9.0.pack",
            "processorName": "cm0",
            "deviceName": "PN7462AU-C3-00",
            "problemMatcher": [],
            "group": {
                "kind": "build",
                "isDefault": true
              }
          }
```

Terminal output with J-Link:
```
*  Executing task: embedded-debug.flash: Flash Device 

Device Selected: Vxx9xCSE
Parse debug description...
Create system description...
Connect to debug unit (CMSIS-DAP)...
Configure debug unit (SWD @ 2000000 Hz)...
Create device map...
Add Flash Algorithm...
        'PN7xxxxx PF (Jun 15 2016)' (0x00203000 - 0x0022c5ff)
Add Flash Algorithm...
        'PN7xxxxx EE (Jun 15 2016)' (0x00201200 - 0x00201fff)
Connect ('haltOnConnect') to device PN7462AU-C3-00...
ERROR: Cannot connect to USB device, it may be already in use, need a firmware update, or need reconnecting (RDDI-ERROR 13)
ERROR: Sequence 'DebugPortSetup' failed
Disconnect from device...
Disconnect from debug unit...
Debug connection failed
Flash failed

 *  The terminal process terminated with exit code: 1. 
```

Same effect with other programmers

kubichris, 2023-04-13:
Update:

I replaced the pack to latest one (Keil.STM32F0xx_DFP.2.1.1.pack), and it works with ST-Link, but with uLink2 have the same error as before:

 ```
*  Executing task: embedded-debug.flash: Flash Device 

Device Selected: 0001A0000000
Parse debug description...
Create system description...
Connect to debug unit (CMSIS-DAP)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Add Flash Algorithm...
        'STM32F0xx 32kB Flash' (0x08000000 - 0x08007fff)
Connect ('haltOnConnect') to device STM32F030K6Tx...
ERROR: Cannot connect to USB device, it may be already in use, need a firmware update, or need reconnecting (RDDI-ERROR 13)
ERROR: Sequence 'DebugPortSetup' failed
Disconnect from device...
Disconnect from debug unit...
Debug connection failed
Flash failed

 *  The terminal process terminated with exit code: 1. 
```

thegecko, 2023-06-21:
@kubichris sorry for the late reply, this issue can occur if something else is using your USB device. Can you confirm if its still an issue?

thegecko, 2023-06-21:
OK, one issue may be that ULINK2 is a HID device which we don't currently support. Let us investigate this further.

vescovd1, 2024-01-03:
I also get the exact same issue. I am wondering just what the error code indicates. 
I almost suspect sometimes the port is still active even though the previous session has ended or something like that.
I too am using the ULINK2 it seems to happen when going back and forth between flash task and debugging. I occasionally can get both to work individually but it does not like when I switch.

jreineckearm, 2024-01-08:
@vescovd1 , could you please also share the relevant parts of your `launch.json` and `task.json` files?

ssrb, 2024-01-09:
Same issue with Muselab Wireless CMSIS-DAP/DAPLink

It works with uvision but not with the debug extension:

[image]

```
"tasks": [
		{
			"type": "embedded-debug.flash",
			"serialNumber": "${command:device-manager.getSerialNumber}",
			"program": "${command:embedded-debug.getApplicationFile}",
			"cmsisPack": "${command:device-manager.getDevicePack}",
			"problemMatcher": [],
			"connectMode": "preReset",
			"resetMode": "hardware",
			"resetAfterConnect": true,
			"label": "embedded-debug.flash: Flash Device"
		}
	]
```

```
Device Selected: 0001A0000002
Parse debug description...
Create system description...
Connect to debug unit (CMSIS-DAP)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Add Flash Algorithm...
        'STM32L4Rx 2MB Dual Bank Flash' (0x08000000 - 0x081fffff)
Connect ('preReset') to device STM32L4R5QIIx...
Pre-connect hardware reset...
ERROR: Cannot connect to USB device, it may be already in use, need a firmware update, or need reconnecting (RDDI-ERROR 13)
DSQ ERROR: DAP access error
ERROR: Sequence 'ResetHardware' failed
Disconnect from device...
Disconnect from debug unit...
Debug connection failed
Flash failed
```

Because of that I still need to use uvision.

saurlax, 2024-01-19:
I am a newcomer to embedded systems and seem to have encountered the same problem.

I work on STM32F103C8T6, using an unknown CMSIS-DAP (maybe called DAPmini)

```
Pack: Keil::STM32F1xx DFP@2.4.1
Device: dapmini
Manufacturer: CMSIS-DAP by ARM
Product: CMSIS-DAP
Serial: 303030314130303030303030
VendorID: 0x483
ProductID: 0x572a
```

[image]

```json
{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "CMSIS Build",
            "type": "cmsis-csolution.build",
            "solution": "${command:cmsis-csolution.getSolutionPath}",
            "buildType": "${command:cmsis-csolution.getBuildType}",
            "targetType": "${command:cmsis-csolution.getTargetType}",
            "problemMatcher": [],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
        {
            "label": "Flash Device",
            "type": "embedded-debug.flash",
            "program": "${command:embedded-debug.getApplicationFile}",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "cmsisPack": "${command:device-manager.getDevicePack}",
            "problemMatcher": [],
            "dependsOn": "CMSIS Build"
        }
    ]
}

```

```
Device Selected: 303030314130303030303030
Parse debug description...
Create system description...
Connect to debug unit (CMSIS-DAP)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Add Flash Algorithm...
        'STM32F10x Med-density Flash' (0x08000000 - 0x0801ffff)
Connect ('haltOnConnect') to device STM32F103C8...
ERROR: Cannot connect to USB device, it may be already in use, need a firmware update, or need reconnecting (RDDI-ERROR 13)
ERROR: Sequence 'DebugPortSetup' failed
Disconnect from device...
Disconnect from debug unit...
Debug connection failed
Flash failed
```

I really want to use Keil Studio. Should I switch to ST-LINK currently?

saurlax, 2024-01-21:
Since buying an ST-LINK v2 and trying it out, I have successfully flashed the program into my stm32 🎉. It seems that currently I need to use ST-LINK instead of CMSIS-DAP.

thegecko, 2024-01-22:
@saurlax I'm glad you now have this working. TBH, I've never used CMSIS-DAP with an ST device, so not sure if your original approach should have worked.

@ssrb I'm not aware of that debug probe, if it supports daplink (and you only need flashing) then the `embedded-debug.daplink-flash` task may work better?

@kubichris @vescovd1 We don't support ulink2 in the embedded debugger so we are intrigued that you got this to work at all. Could you share your launch config?
As @jreineckearm mentions, we have a preview out now of the arm debugger which does support this debug probe.

kubichris, 2024-01-29:
Sorry for late answer.
Now, we in another project, but same results with ulink2 as a CMSIS-DAP
device:

[image: 338cb406-60d5-4aa3-8cce-100b1a1869c0.png]
So, the ulink firmwared as CMSIS-DAP adapter.

Configuration:

        {
            "label": "embedded-debug.flash: Flash Device",
            "type": "embedded-debug.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${workspaceFolder}/build/ePaper.bin",
            "cmsisPack":
"${workspaceFolder}/Pack/Keil.STM32U5xx_DFP.2.0.3.pack",
            "deviceName": "STM32U575AIIx",
            "problemMatcher": [],
            "group": {
                "kind": "build"
            }
        }

Result (:

> Device Selected: V2291CSE
> Parse debug description...
> Create system description...
> Connect to debug unit (CMSIS-DAP)...
> Configure debug unit (SWD @ 10000000 Hz)...
> Create device map...
> Add Flash Algorithm...
>         'STM32U5xx 2M NSecure Flash' (0x08000000 - 0x081fffff)
> Add Flash Algorithm...
>         'STM32U5xx 2M Secure Flash' (0x0c000000 - 0x0c1fffff)
> Connect ('haltOnConnect') to device STM32U575AIIx...
> ERROR: Cannot connect to USB device, it may be already in use, need a
> firmware update, or need reconnecting (RDDI-ERROR 13)
> ERROR: Sequence 'DebugPortSetup' failed
> Disconnect from device...
> Disconnect from debug unit...
> Debug connection failed
> Flash failed

Chris.

Rob Moran ***@***.***> ezt írta (időpont: 2024. jan. 22., H,
18:18):

> @saurlax <https://github.com/saurlax> I'm glad you now have this working.
> TBH, I've never used CMSIS-DAP with an ST device, so not sure if your
> original approach should have worked.
>
> @ssrb [...]

kubichris, 2024-02-02:
> @kubichris <https://github.com/kubichris> could you link me to the
> CMSIS-DAP firmware you are using on your ULINK device?
>
> Thanks!

Yes, I used the ARM's official bootmode utility:
https://developer.arm.com/documentation/101455/0100/Appendix/Utilities/Boot-Mode-Utility?lang=en

Rob Moran ***@***.***> ezt írta (időpont: 2024. febr. 2., Pén
20:08):

jreineckearm, 2024-02-03:
Thanks for sharing more information, @kubichris !

The ULINK2 CMSIS-DAP firmware implementation is based on CMSIS-DAPv1 and exposes the debugger as an HID device. For the Embedded Debugger, we require a CMSIS-DAPv2 firmware implementation which uses a different USB device class. Hence, we currently neither support native ULINK2, nor ULINK2 in CMSIS-DAP mode. The same is true for other CMSIS-DAPv1 HID implementations. Which I suspect is the reason for others in this thread failing with their CMSIS-DAP probes. But I cannot tell for sure as I don't have those probes to test it. I appreciate that the Embedded Debugger needs enhancements to better guide you through these limitations.

Unfortunately, there are currently no plans to update the ULINK2 CMSIS-DAP implementation to CMSIS-DAPv2.
An alternative to get you going with ULINK2 and other CMSIS-DAPv1 HID devices is to switch to the Arm Debugger extension which currently is at preview quality: https://marketplace.visualstudio.com/items?itemName=Arm.arm-debugger
Would it be an option for you to try this out?

kubichris, 2024-02-03:
Thank you @jreineckearm !

On Monday I will try this extension on my project (STM32U585 MCU) with uLink2. 
Should I reset uLINK from CMSIS-DAP or should it work in either mode?

jreineckearm, 2024-02-05:
@kubichris , please keep ULINK2 in CMSIS-DAP boot mode. The Arm Debugger extension also uses CMSIS-DAP for ULINK2 and supports CMSIS-DAPv1 (HID). Tools that support the native ULINK2 mode, like uVision, can dynamically switch to it. Hence, would still work with this setup.

kubichris, 2024-02-05:
Hi!

I tried it, but it wasn't easy because of the license activation. The
activation popup only shows after restarting vsCode.

So, I had an error :

Executing task: arm-debugger.flash: Flash Device
ERROR(CLI223): No CMSIS-Packs found, CMSIS-Packs must be installed using
the CMSIS-Pack Manager.
Flash failed

tasks.json
        {
            "type": "arm-debugger.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${workspaceFolder}/release/ePaper.bin",
            "cmsisPack":
"${workspaceFolder}/Pack/Keil.STM32U5xx_DFP.2.0.3.pack",
            "deviceName": "STM32U575AIIx",
            "problemMatcher": [],
            "label": "arm-debugger.flash: Flash Device"
        }

launch.json
        {
            "name": "Arm Debugger",
            "type": "arm-debugger",
            "request": "launch",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${workspaceFolder}/release/ePaper.bin",
            "cmsisPack":
"${workspaceFolder}/Pack/Keil.STM32U5xx_DFP.2.0.3.pack",
            "deviceName": "STM32U575AIIx",
        }

I know that error, but I don't want to install this utility. Usally I copy
the pack into the project's directory, and it works with embedded-debug :
        {
            "label": "embedded-debug.flash: Flash Device",
            "type": "embedded-debug.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${workspaceFolder}/build/ePaper.bin",
            "cmsisPack":
"${workspaceFolder}/Pack/Keil.STM32U5xx_DFP.2.0.3.pack",
            "deviceName": "STM32U575AIIx",
            "problemMatcher": [],
            "group": {
                "kind": "build"
            }
        },

Jens Reinecke ***@***.***> ezt írta (időpont: 2024. febr. 5.,
Hét 11:32):