ARM-software/vscode-embedded-debug #17: Use with Arm MPS2 development board?

DavidA2014, 2023-04-17:
Hi, can you please offer me any guidance on how to use this extension to debug a Cortex M4 elf file, built using armclang, on an Arm MPS2 board, using the CMSIS-DAP USB interface?

DavidA2014, 2023-04-20:
Thanks for your answer. Do I need any other specific extensions besides ‘Arm Embedded Debugger’ for debugging?

Do I need OpenOCD and / or gdb?

I guess your extension offers similar functionality to the 'Cortex-Debug' extension?

thegecko, 2023-04-20:
> Do I need any other specific extensions besides ‘Arm Embedded Debugger’ for debugging?

No, although the extension is included with our other tooling in the [Keil Studio Extension Pack](https://marketplace.visualstudio.com/items?itemName=Arm.keil-studio-pack)

> Do I need OpenOCD and / or gdb?

No, a basic step debugger is included in the extension

> I guess your extension offers similar functionality to the 'Cortex-Debug' extension?

Our pack includes the SVD viewer originally developed as part of Cortex Debug, but doesn't yet include memory viewing.

The biggest difference however is that this debugger works natively in a browser environment (e.g. vscode.dev or GitHub codespaces).

DavidA2014, 2023-04-21:
Thanks for your answer. Please will you tell me what is the appropriate way to create launch.json in this context?

thegecko, 2023-04-21:
> Please will you tell me what is the appropriate way to create launch.json in this context?

In which context?

DavidA2014, 2023-04-21:
As I understand it, if using the VS Code debugger, there is a 'create a launch.json file link' in the Run start view. (Ref. [Launch configurations](https://code.visualstudio.com/Docs/editor/debugging#_launch-configurations)).

How does one create launch.json  if using the ‘Arm Embedded Debugger’ extension? (That's what I meant by this context).

thegecko, 2023-04-21:
It's specified as outlined in my post above. The debugger used is set by the type:

```
{
      ...
      "type": "embedded-debug",
      ...
}
```

DavidA2014, 2023-04-25:
@thegecko I would be grateful for some more guidance please. I really don't understand the workflow required for what I want to do.

I want to develop Cortex-M4 code on an Arm MPS2 dev. board.  I have chosen to use the **startup_Cortex-M4_AC6** project, supplied with Arm Development Studio, as an example to get started.

One way of running code on the MPS2 is to copy the elf file to the board's flash drive (which appears as a USB drive in Windows) and to specify that elf file in images.txt. When rebooted, the elf file should load and execute, but in my case it doesn't. I guess this is not the right place to discuss that problem, but it leads me on to trying to load the elf using the Arm Embedded Debugger extension instead.

I want to connect to the MPS2 using the CMSIS-DAP USB interface, so I assume I should create a **embedded-debug.daplink-flash:Flash Device (DAPLink)** task. Is that correct?

Here is my task:

    {
    	"version": "2.0.0",
    	"tasks": [
    		{
    			"type": "embedded-debug.daplink-flash",
    			"serialNumber": "${command:device-manager.getSerialNumber}",
    			"program": "/Users/daldrich/Development Studio Workspace/startup_Cortex-M4_AC6/startup_Cortex-M4_AC6.axf",
    			"problemMatcher": [],
    			"label": "embedded-debug.daplink-flash: Flash Device (DAPLink)"
    		}
    	]
    }

When I run that task will it program the elf into Flash on the MPS2?

Will that overwrite the existing boot loader, or whatever is there, that loads an elf from the Flash drive?

In other words, is this the correct approach?

DavidA2014, 2023-04-26:
Thanks. Does the CMSIS-DAP USB interface use DAPLink?

thegecko, 2023-04-27:
CMSIS-DAP is a protocol, DAPLink is firmware which supports CMSIS-DAP.

DAPLink generally supports drag-n-drop flashing (Mass storage device), CMSIS-DAP compliant targets don't have to.

thegecko, 2023-06-21:
@DavidA2014 Were you able to get this working?

closed 2023-07-03 by DavidA2014