ARM-software/vscode-embedded-debug #19: Cannot downloaded firmware to STM32L475VET6 using stlink

DinoHaw, 2023-07-17:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11
- VS Code Version: 1.77.3
- Extension Version: v1.0.6
- Target Device: STM32L475VET6
- Keil Studio Pack (10 extension packages)
- Cannot downloaded firmware to STM32L475VET6 using stlink

**To Reproduce**
Steps to reproduce the behavior:
1. using csolution and "Build"
2. build finished successfully!
3. click “Debug”
4. flash failed and see the following information

**The following is terminal information**
Device Selected: 066AFF313933554D43221330
Parse debug description...
Create system description...
Connect to debug unit (ST-Link)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Add Flash Algorithm...
        'STM32L4xx 512 KB Flash' (0x08000000 - 0x0807ffff)
Connect ('underReset') to device STM32L475VETx...
Assert hardware reset...
ERROR: General error on memory or register access. (RDDI-ERROR 22)
ERROR: AP register read failed (@0x000000f8)
ERROR: AP register read failed (@0x000000f8)
DSQ ERROR: Debug access failed - target access failed
ERROR: Sequence 'DebugDeviceUnlock' failed
Disconnect from device...
Disconnect from debug unit...
Debug connection failed
Flash failed

thegecko, 2023-07-17:
cc @jreineckearm

jreineckearm, 2023-07-20:
@DinoHaw , thank you for the report.

Are you running the latest ST-Link FW version?
Are you using a software project that is publicly available?
Does the issue happen on every attempt or has it intermittent character?