ARM-software/vscode-embedded-debug #22: Wrong DFP pack used when flashing the board

VVESTM, 2023-09-13:
Type: Bug Report

I can compile my project with VSCode extensions but I have some issue while flashing. As seen on this capture, the pack used to flash is not the one I have specified in task.json file.

[image]

**information on my environment:**
- OS and Version: Windows
- VS Code Version: 1.82.0
- Extension Version: Keil Studio Pack 1.10.0
- Target Device:STM32 NUCLEO-U575ZI-Q

**Steps to reproduce the behavior:**
1. Go to 'CMSIS extension', select the .csolution.yml file
2. Build the project
3. Connect the board (I can see it in the device manager)
4. Flash the board with "Run" button. Here, we get the error previously described (wrong pack is used)

thegecko, 2023-09-15:
Thanks for raising this @VVESTM 

To help us investigate, would you mind sharing the debug output when you undertake this action?

To do so, change the logging verbosity setting of the Embedded Debug extension to `debug`

[image]

and open the `Arm Debugger Extension` Output channel.

[image]

Debug information should then be streamed out in this window as you undertake a debug connection.

Thanks!

thegecko, 2023-09-15:
@VVESTM I think you may be seeing 2 issues here as there is no link between the cmsis-csolution and embedded-debug extensions.

Regarding the PDSC file not found. it could be that it doesn't exist (please check the file `/dfp/stm32u5xx/STMicroelectronics.stm32u5xx_dfp.pdsc` exists in your workspace).

@mcgordonite do you have any idea where the popup error is originating?

VVESTM, 2023-09-15:
I have printed the content of ${workspaceFolder} using the keybinding method described [here](https://stackoverflow.com/questions/53485115/how-to-display-current-values-of-vs-codes-predefined-variables-such-as-work).
It corresponds to my code and I can see the pdsc and FLM files by performing a ls on its.

mcgordonite, 2023-09-18:
I think there are two problems here:
1. The Embedded Debug extension is asking the CMSIS extension to install the STM32U5 DFP even though it is already present. I don't think it should try installing the pack in this case.
2. The CMSIS extension is failing to download the pack because it thinks version 2.1.0 doesn't exist. It looks like version 2.1.0 was published in March so this shouldn't be happening.

To help investigate number 2, can you check if there are any "missing pack" errors reported in the Csolution file @VVESTM? They should appear in the problems panel or as red squiggly highlighting in the file. If there are problems reported, does installing the pack work from there?

thegecko, 2023-09-18:
@mcgordonite I believe packs in use are local, so the download failure isn't a surprise.

The main issue here is the `Run` and `Debug` convenience buttons are not picking up the tasks and debug configurations correctly - we will look into this.

As a workaround, tasks can be run using the Menu: `Terminal > Run Task` and debug can be started using the debug panel (`Play` button to the left)

VVESTM, 2023-09-18:
Thanks for this workaround @thegecko. I confirm this is working on my side.
 Of course, keep us posted if any updates... 
Thanks a lot for your support.

thegecko, 2023-09-20:
I believe this issue is resolved, feel free to re-open it if this is not the case.

closed 2023-09-20 by thegecko