ARM-software/vscode-embedded-debug #23: use DAP-LINK

Achiyun, 2023-09-15:
*  正在执行任务: embedded-debug.flash: Flash Device 

Device Selected: 378E9C348EFD6A1AEDCA2F807DC96338
Only absolute URLs are supported
Flash failed

thegecko, 2023-09-15:
Thanks for the feedback, can you provide some more detail, please?

e.g. what does your configuration look like?

thegecko, 2023-09-16:
I can see the same issue and believe it has been triggered by the new VS Code update. I've raised the issue here:

https://github.com/microsoft/vscode/issues/193239

An update with a fix is being worked on and should be released soon.

Achiyun, 2023-09-16:
> 我可以看到同样的问题，并相信它是由新的 VS Code 更新触发的。我在这里提出了这个问题：
> 
> [微软/vscode#193239](https://github.com/microsoft/vscode/issues/193239)
> 
> 正在开发包含修复程序的更新，应该很快就会发布。

thanks

thegecko, 2023-09-17:
A fix has been deployed in the latest release (v1.0.10), please confirm if this has solved your issue.

Achiyun, 2023-09-29:
> 最新版本（v1.0.10）中已部署修复程序，请确认这是否解决了您的问题。

resolved

closed 2023-09-29 by thegecko