ARM-software/vscode-embedded-debug #24: Error Path

Achiyun, 2023-09-16:
When I click on 'run' to download, it will prompt that I cannot find the axf file. For MDK6 I am using in wsl, it will search for the. axf file in the directory of 'win', such as' c:/home/chxx/workspace/', instead of searching in wsl

pozitp, 2023-11-29:
> When I click on 'run' to download, it will prompt that I cannot find the axf file. For MDK6 I am using in wsl, it will search for the. axf file in the directory of 'win', such as' c:/home/chxx/workspace/', instead of searching in wsl

did you fixed it? i have this error too

thegecko, 2023-11-29:
Hi @Achiyun @pozitp 

This setup isn't an environment we have tested or support, however we are keen to see if we can resolve the issue.

To confirm, you are using WSL on Windows to use a linux VM. You are connecting to this VM using VS Code on your host and flashing from there and expect the extension to pick up files from the linux container and not the windows host.

pozitp, 2023-11-29:
> Hi @Achiyun @pozitp
>
> This setup isn't an environment we have tested or support, however we are keen to see if we can resolve the issue.
>
> To confirm, you are using WSL on Windows to use a linux VM. [...]

Yes, I use WSL on Windows to use a Linux VM (WSL2, Ubuntu LTS). I am connecting through VSCode to VM, and flashing from there. I except pick up files from the linux container.

Achiyun, 2023-11-29:
> Yes, I use WSL on Windows to use a Linux VM (WSL2, Ubuntu LTS). I am connecting through VSCode to VM, and flashing from there. I except pick up files from the linux container.

NO

thegecko, 2023-11-30:
Thanks for your feedback on this issue. We have identified the problem and a fix requires a considerable architecture change. Please bear with us as we look to schedule this.

Achiyun, 2023-12-01:
> Thanks for your feedback on this issue. We have identified the problem and a fix requires a considerable architecture change. Please bear with us as we look to schedule this.

Thanks

thegecko, 2024-04-25:
Update: the latest release has been re-worked so that it should be possible to use it in a WSL or SSH-remote environment (since v1.1.0)