ARM-software/vscode-embedded-debug #29: Does it support M55?

ccli0, 2024-03-07:
Hi,

I used the embedded-debug extension to debug the M55 device, but I encountered the following error.
What could be the problem?

**Use debug button error log**

Device Selected: 13010000000A25020450395D00002A75
Parse debug description...
Create system description...
Connect to debug unit (CMSIS-DAP)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Connect ('haltOnConnect') to device M55M1H2LJAE...
ERROR: RDDI device 3 - No connection to target. (RDDI-ERROR 57)
Disconnect from device...
Disconnect from debug unit...
ERROR: Debug connection failed

launch.json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Embedded Debug",
            "type": "embedded-debug",
            "request": "launch",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${command:embedded-debug.getApplicationFile}",
            "cmsisPack": "${command:device-manager.getDevicePack}",
            "deviceName": "${command:device-manager.getDeviceName}",
            "debugFrom": "main"
        }
    ]
}

tasks.json
{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "CMSIS Build",
            "type": "cmsis-csolution.build",
            "solution": "${command:cmsis-csolution.getSolutionPath}",
            "problemMatcher": [],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
        {
            "label": "Flash Device",
            "type": "embedded-debug.flash",
            "program": "${command:embedded-debug.getApplicationFile}",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "cmsisPack": "${command:device-manager.getDevicePack}",
            "deviceName": "${command:device-manager.getDeviceName}",
            "problemMatcher": [],
            "verifyFlash": false,
            "dependsOn": "CMSIS Build"
        },
    ]
}

**Use run button error log**

 *  Executing task: Flash Device 
Device Selected: 13010000000A25020450395D00002A75
Parse debug description...
Create system description...
Connect to debug unit (CMSIS-DAP)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Add Flash Algorithm...
        'M55M1 APROM 2MB' (0x00100000 - 0x002fffff)
Connect ('haltOnConnect') to device M55M1H2LJAE...
ERROR: RDDI device 3 - No connection to target. (RDDI-ERROR 57)
Disconnect from device...
Disconnect from debug unit...
Debug connection failed
Flash failed

jreineckearm, 2024-03-08:
Thanks, @ccli0 , for raising this!

I see that you are using a Nuvoton M55M1H2LJAE which we don't have access to yet. So, I unfortunately cannot check on our end.

It could be a side effect of on-chip logic which requires specialties during the debug connection. I see that Nuvoton include this device in the `NuMicro_DFP` pack released last week. But without CMSIS debug descriptions. It would be worth to check if something is needed there: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/coresight_setup.html

An alternative could be to give it a try with the recently released Arm Debugger extension: https://marketplace.visualstudio.com/items?itemName=Arm.arm-debugger
But I don't expect it to work better if CMSIS debug descriptions are required, for example to unlock the chip.

jreineckearm, 2024-04-03:
@ccli0 , apologies for taking a while with this. We've got in the meanwhile access to the device.

It turns out that the system is ADIv6 based. Hence, needs more information in the DFP's PDSC file to work correctly with debuggers which do not do a full system auto-detection on every debug connection. Like the Arm Embedded Debugger.

You at minimum would need to add an [accessportV2](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_accessportV2) element for the concerned device. And an according `__apid` reference in the device's [debug](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_debug) element.

Could you please give this a try?

ccli0, 2024-05-23:
@jreineckearm,
I would like to confirm something. What operation does the resetRun option in the task.json perform in VSCode?
[image]

For the M55M1, when resetRun is set to true, an error occurs if you try to download the code a second time or enter debug mode after the initial download is completed. When using Keil MDK, the following message appears:
[image]

Could you provide some suggestions?

closed 2024-06-12 by ccli0