ARM-software/vscode-embedded-debug #30: Unable to Debug

Kbdman, 2024-04-13:
Type: Bug Report

**Describe the bug**
- OS and Version:  Ubuntu 23.10
- VS Code Version: 1.88.1
- Extension Version: v1.0.13
- Target Device: STLinkV2+STM32F407ZGT6
- Other extensions you installed (and if the issue persists after disabling them):
- A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash my program with a task ,flash success
    ```
    {
      "type": "embedded-debug.flash",
      "serialNumber": "${command:device-manager.getSerialNumber}",
      "program": "${command:embedded-debug.getApplicationFile}",
      "cmsisPack": "${command:cmsis-csolution.getTargetPack}",
      "problemMatcher": [],
      "label": "embedded-debug.flash: Flash Device",
      "deviceName": "STM32F407ZGTx"
    }
    ```

2. Go to 'CMSIS' UI
3. Click on the debug button. My debug task is :
    ```
      {
      "name": "Embedded Debug",
      "type": "embedded-debug",
      "request": "launch",
      "serialNumber": "${command:device-manager.getSerialNumber}",
      "program": "${command:embedded-debug.getApplicationFile}",
      "cmsisPack": "${command:device-manager.getDevicePack}",
      "debugFrom": "main",
      "deviceName": "STM32F407ZGTx"
    }
    ```
4. failed to start the debug .the error log
```
Device Selected: 52ff70064852854829401167
Parse debug description...
Create system description...
Connect to debug unit (ST-Link)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Parse dbgconf file...
Connect ('underReset') to device STM32F407ZGTx...
Assert hardware reset...
Device connected
Hardware reset de-asserted
Device halted after connect
Reset ('auto')...
Reset completed
Load application...
ERROR: Offset is outside the bounds of the DataView
Disconnect from device...
Disconnect from debug unit...
```
After the log "Loading application", there is an error log "ERROR: Offset is outside the bounds of the DataView", and then the debugger stopped

thegecko, 2024-04-25:
Thanks for raising this issue, I can't see any obvious issues with your configuration.

@mcgordonite @jreineckearm any ideas?

In the meantime, we have recently released the [Arm Debugger extension](https://marketplace.visualstudio.com/items?itemName=Arm.arm-debugger) which is a more professional debugger and may work better for you.

jreineckearm, 2024-04-25:
Given the point in time and the error message, this could point to a defect in our ELF/DWARF loader.
@Kbdman , could you please share more information on the application you work with?
Is it publicly available so that we can try to reproduce?
Which compiler toolchain are you using? This would have an impact on the debug info generated into the loaded file.

Kbdman, 2024-05-15:
Sorry for late response, I haven't try it for several weeks.Seems my extension version comes to 1.1.2.
The error logs changes to 
```
Arm Embedded Debugger
v1.1.2
Arm
arm.com
41,812
(1)
Debugger for Arm Cortex-M based microcontrollers (MCUs), development boards and debug probes, implementing the Microsoft Debug Adapter Protocol (DAP)
Disable
Uninstall
This extension is enabled globally.

DETAILS
FEATURES
CHANGELOG
EXTENSION PACK
PROBLEMS
2
OUTPUT
DEBUG CONSOLE
TERMINAL
PORTS

Device Selected: 52ff70064852854829401167
Parse debug description...
Create system description...
Connect to debug unit (ST-Link)...
Configure debug unit (SWD @ 10000000 Hz)...
Create device map...
Parse dbgconf file...
Connect ('underReset') to device STM32F407ZGTx...
Assert hardware reset...
Device connected
Hardware reset de-asserted
Device halted after connect
Reset ('auto')...
Reset completed
Load application...
ERROR: Error: Compilation unit with unsupported DWARF v5
Disconnect from device...
Disconnect from debug unit...
```
My toolchain is
```
➜  T1 /home/lihai/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=/home/lihai/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/home/lihai/.vcpkg/artifacts/2139c4c6/compilers.arm.arm.none.eabi.gcc/13.2.1/bin/../libexec/gcc/arm-none-eabi/13.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: /data/jenkins/workspace/GNU-toolchain/arm-13/src/gcc/configure --target=arm-none-eabi --prefix=/data/jenkins/workspace/GNU-toolchain/arm-13/build-arm-none-eabi/install --with-gmp=/data/jenkins/workspace/GNU-toolchain/arm-13/build-arm-none-eabi/host-tools --with-mpfr=/data/jenkins/workspace/GNU-toolchain/arm-13/build-arm-none-eabi/host-tools --with-mpc=/data/jenkins/workspace/GNU-toolchain/arm-13/build-arm-none-eabi/host-tools --with-isl=/data/jenkins/workspace/GNU-toolchain/arm-13/build-arm-none-eabi/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-gnu-as --with-headers=yes --with-gnu-ld --with-native-system-header-dir=/include --with-sysroot=/data/jenkins/workspace/GNU-toolchain/arm-13/build-arm-none-eabi/install/arm-none-eabi --with-multilib-list=aprofile,rmprofile --with-pkgversion='Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)' --with-bugurl=https://bugs.linaro.org/
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.1 20231009 (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 
```

Kbdman, 2024-05-15:
After setting dwarf version to dwarfv4 by compiling flags, everything works

thegecko, 2024-05-15:
> After setting dwarf version to dwarfv4 by compiling flags, everything works

This is great news. We hope to add dwarf 5 support sometime in future

closed 2024-05-15 by Kbdman