ARM-software/vscode-embedded-debug #31: resetRun option in the task.json perform in VSCode (M55 device)

ccli0, 2024-06-12:
I would like to confirm something. What operation does the resetRun option in the task.json perform in VSCode?
[image]
For the M55M1, when resetRun is set to true, an error occurs if you try to download the code a second time or enter debug mode after the initial download is completed. When using Keil MDK, the following message appears:
[image]
Could you provide some suggestions?

thegecko, 2024-06-12:
It should undertake a hardware reset after the flashing completes, but probably depends on the target.

ccli0, 2024-06-14:
Hi @thegecko,
Can we add a customized reset process when resetRun option set to true or catch exceptions when the reset fails?